sirleech/Webduino #63: GarageDoorOpener.ino not compiling - Arduino 1.0.6

Sudharsp, 2015-01-18:
Hi ,
I am new to Ardiuno. Getting following error during compiling GarageDoorOpener.ino

 In file included from Webserver.cpp:4:
/Webduino.h: In member function 'int WebServer::read()':
Webduino.h:729: error: ambiguous overload for 'operator==' in '((WebServer*)this)->WebServer::m_client == 0'
/Webduino.h:729: note: candidates are: operator==(int, int) <built-in>
C:\Program Files (x86)\Arduino\libraries\Ethernet/EthernetClient.h:27: note:                 virtual bool EthernetClient::operator==(const EthernetClient&)

unwiredben, 2015-01-19:
Which version of the Arduino IDE are you using?  This library was developed before the days of Arduino 1.0, and may have broken with recent Ethernet library changes.

Sudharsp, 2015-01-21:
Thanks for your response. I am using Arduino 1.0.6.
This is my first project and I have very limited programming knowledge.
Any help would be appreciated.

Sudharsp, 2015-01-21:
Thanks for your response. I am using Arduino 1.0.6.
 
This is my first project and I have very limited programming knowledge.
 
Any help would be appreciated.
 
Regards...Sudhar
 
Date: Mon, 19 Jan 2015 12:13:54 -0800