sirleech/Webduino #67: not works on adruino ide 1.6.5, compilation error

taoloo897, 2015-07-09:
Hi again my friend.like we talked on email i put the problems here for arduino ide 1.6.5
If you can make some modification for it will be greate.
Many thanks

> file included from C:\Program Files
> (x86)\Arduino\libraries\Webduino/WebServer.h:31:0,
>
>                  from Imagini.h:1,
>
>                  from casa.ino:6:
>
> C:\Program Files 
> (x86)\Arduino\libraries\Ethernet\src/EthernetClient.h: In member function 'int WebServer::read()':
>
> C:\Program Files
> (x86)\Arduino\libraries\Ethernet\src/EthernetClient.h:27:16: note: 
> candidate
> 1: virtual bool EthernetClient::operator==(bool)
>
>    virtual bool operator==(const bool value) { return bool() == value; 
> }
>
>                 ^
>
> In file included from c:\program files 
> (x86)\arduino\hardware\tools\avr\avr\include\avr\eeprom.h:570:0,
>
>                  from C:\Program Files 
> (x86)\Arduino\hardware\arduino\avr\cores\arduino/USBAPI.h:25,
>
>                  from C:\Program Files 
> (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:225,
>
>                  from C:\Program Files 
> (x86)\Arduino\libraries\Ethernet\src/EthernetClient.h:3,
>
>                  from C:\Program Files 
> (x86)\Arduino\libraries\Webduino/WebServer.h:31,
>
>                  from Imagini.h:1,
>
>                  from casa.ino:6:
>
> C:\Program Files (x86)\Arduino\libraries\Webduino/WebServer.h:728:19: note:
> candidate 2: operator==(int, int) <built-in>
>
>    if (m_client == NULL)
>

agnunez, 2015-07-13:
Hi,

I found similar errors with arduino-nightly. My code was working until now with Webduino. If I install Webduino in ~/sketchbook/libraries/ as usual, it now gives:

fatal error: WebServer.h: No such file or directory. compilation terminated.Error compiling.

I've tried also in ~/arduino-nightly/libraries/ but then the errors are similars to aboce (trying to use Ethernet library instead):

~/arduino-nightly/libraries/Ethernet/src/EthernetClient.cpp:175:1: error: expected unqualified-id before '+' token
 +uint8_t EthernetClient::getSocketNumber () {
 ^
Error compiling.

I am using latest version of Webduino at github.

Thanks,
Agustin

q2dg, 2015-08-27:
If tried to download the zip from Github, rename it from "Webduino-master" to "Webduino" and install it in the IDE (1.6.5) via "Add ZIP library" option. Then, I've tried to compile https://github.com/sirleech/Webduino/blob/master/examples/Web_Demo/Web_Demo.ino . Zero problems. I'm using Fedora 22 64 bits