sirleech/Webduino #69: index.htm try to load a page from sd card

NitrofMtl, 2015-09-29:
Hi.

 I'm just begun to use this lib, and I try to fit it with a sketch a have already begun. In that on, my page template is store on the sd card. I do not understand all the example because I'm a newbie in programming in both web page and c++.

I've tried to modify the example and It compile but the page only respond with a bunch on html call text:

0HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-Allow-Origin: * Content-Type: text/html; charset=utf-8 33HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-Allow-Origin: * Content-Type: text/html; charset=utf-8 68HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-Allow-Origin: * Content-Type: text/html; charset=utf-8 79HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-Allow-Origin: * Content-Type: text/html; charset=utf-8 67HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-Allow-Origin: * Content-Type: text/html; charset=utf-8 84HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-Allow-Origin: * Content-Type: text/html; charset=utf-8 89HTTP/1.0 200 OK Server: Webduino/1.7 Access-Control-

here what I've done so far:

void outputPins(WebServer &server, WebServer::ConnectionType type, bool addControls = false) // this function should be replace by loadhtml
{

              // web page request
            // send web page
            webFile = SD.open("index.htm");        // open web page file
            if (webFile) {
              while (webFile.available()) {
                server.httpSuccess();
                server.print(webFile.read()); // send web page to client
              }
              webFile.close();
            }
          
  }

but here is the idea: can it be possible to include file template or file get request on it.. that could very simplify both html and c++ coding.... (so I think so... :P)

and also: Is client post method support nested arrays ? 

thanks

unwiredben, 2015-09-29:
I do not recommend using SD library with this.  Both the SD card and the WS5100 use the SPI connection and it's possible that accessing SD card can interfere with the network connection.

closed 2015-09-29 by unwiredben

systronix, 2015-09-30:
But why not, if the SPIs have different slave selects? They are completely separate in that case.

unwiredben, 2015-09-30:
When I originally wrote this library, the SD card slot on the Ethernet shield wasn't working well and may have been wired incorrectly.  This looks to have been addressed with later revisions.

I also think the original Ethernet library wasn't so careful about deasserting chip select between operations.

Anyway, this library is no longer being actively maintained; I accept pull requests, but I don't have a use for this myself anymore.