sirleech/Webduino #71: W5500 compatibility

agnunez, 2016-02-23:
Hi,
Using Webduino for years now without issue, but there was one UNO with hw problem and was replaced by a new one with W5500 ethernet shield, and Webduino does not work. Any news about Webduino compatibility with Wiznet W5500 ethernet chip/shield?
Thanks

unwiredben, 2016-02-24:
I've no personal experience with the W5500 and haven't really worked on this code for a few years.  I'd welcome a compatibility patch, but don't plan on doing that myself.

t30, 2016-06-27:
I solved removing this line:
https://github.com/sirleech/Webduino/blob/master/WebServer.h#L32

You will need to include Ethernet.h or Ethernet2.h in your sketch, see below example:
```
#if defined BUILD_COMM_ETH  
  #define WEBDUINO_FAVICON_DATA "" // no favicon
  #include <SPI.h>
  #include "avr/pgmspace.h" // new include
  #include <Ethernet.h>
//  #include <Ethernet2.h>
  #include "WebServer.h"
#endif
```