sirleech/Webduino #74: Problem compiling with template <class T>

d3iu, 2017-01-17:
exit status 1
'T' has not been declared

i tried declaring it as an INT T; but does not work... the error is in this line 

template<class T> // no-cost stream operator as described at http://arduiniana.org/libraries/streaming/
inline Print &operator <<(Print &obj, T arg)
{
 obj.print(arg);
 return obj;
}

HOW TO FIX IT ?

unwiredben, 2017-01-17:
In the code you posted, it's missing the "<class T>" after the template keyword.  This is in the original file at https://github.com/sirleech/Webduino/blob/master/examples/Web_Demo/Web_Demo.ino

closed 2017-01-17 by unwiredben

d3iu, 2017-01-18:
Found the problem, the line must be in 1 row instead of multiple lines... that`s the error in the code, now it`s working