sirleech/Webduino #77: Need to update Webduino library to Arduino IDE 1.5: Library specification

forseyra, 2017-07-22:
This library needs to be updated to the Arduino IDE 1.5: Library specification. This would make any updates and bug fixes available to the user community via the Arduino Library Manager.

unwiredben, 2017-07-23:
Feel free to submit a pull request. We're not actively maintaining this
code now, I've not personally used it in years

per1234, 2018-02-10:
The requirements and process for adding a library to the Library Manager index is documented here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
Note that it's not necessary to make a full conversion to the 1.5 library format.

Necessary actions:

1. Add library.properties file to the root of the repository. Format is documented here: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
2. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in library.properties.
3. Submit an issue to the [arduino/Arduino issue tracker](https://github.com/arduino/Arduino/issues) that requests the library to be added to the Library Manager index.

In this case, the primary benefit is to allow people to discover and install the library via the Arduino IDE's **Sketch > Include Library > Manage Libraries** feature. Library Manager also provides update notifications whenever a new version of the library is released.