sirleech/Webduino #78: Convert png image

BorgMcz, 2018-07-05:
Please help, I converted the png image using the directfb-csource utility, but the result is a mixture of octal and characters. Can you please advise me how to get the output in HEX.
Thank you.

  "\211PNG\15\12\32\12\0\0\0\15IHDR\0\0\0\20\0\0\0\20\10\2\0\0\0\220\221"
  "h6\0\0\0(tEXtCreation\40Time\0\350t\40""5\40""7\40""2018\40""13:47:0"
  "3\40+0100\302\347D\343\0\0\0\7tIME\7\342\7\5\21/\7\203:X\202\0\0\0\11"
  "pHYs\0\0\13\22\0\0\13\22\1\322\335~\374\0\0\0\4gAMA\0\0\261\217\13\374"
  "a\5\0\0\0PIDATx\332c\374\377\237\201$\300\204K\202\221\21\204\210\325"
  "\0T\12\264\31\2100\365\340\264\201d'\15\234\6\26\2342\214\330\205\7\304"
  "\17\240H\200\261\3773\240G\5\23\36\325X\365\320\337\17\240\364\203\346"
  "H\260\40\202\2135y#;\32M\1\0\17!\22$\351\33[\220\0\0\0\0IEND\256B`\202"

unwiredben, 2018-07-05:
This is still a valid C string and can be used. AFAIK, there's no command-line option for directfb-csource to tell it to use hex, it looks like it uses octal notation because it ends up being more compact due to \0 only taking two characters instead of three.  The largest octal byte is four characters, same as  the largest hex byte (\377 vs \xFF)

closed 2018-07-05 by unwiredben

BorgMcz, 2018-07-06:
Yes I know it is the right output, but I'm looking for a way to use it in the write for arduino or to convert to hex to be used as in the example "Web_Image.ino" Is there any way?

unwiredben, 2018-07-06:
You can write

```
P(ledData) = 
"\211PNG\15\12\32\12\0\0\0\15IHDR\0\0\0\20\0\0\0\20\10\2\0\0\0\220\221"
"h6\0\0\0(tEXtCreation\40Time\0\350t\40""5\40""7\40""2018\40""13:47:0"
"3\40+0100\302\347D\343\0\0\0\7tIME\7\342\7\5\21/\7\203:X\202\0\0\0\11"
"pHYs\0\0\13\22\0\0\13\22\1\322\335~\374\0\0\0\4gAMA\0\0\261\217\13\374"
"a\5\0\0\0PIDATx\332c\374\377\237\201$\300\204K\202\221\21\204\210\325"
"\0T\12\264\31\2100\365\340\264\201d'\15\234\6\26\2342\214\330\205\7\304"
"\17\240H\200\261\3773\240G\5\23\36\325X\365\320\337\17\240\364\203\346"
"H\260\40\202\2135y#;\32M\1\0\17!\22$\351\33[\220\0\0\0\0IEND\256B`\202";
```

and the code will work.  The P macro defines a in-flash char array, and those strings
are valid data for the C++ compiler to initialize the array.