sirleech/Webduino #79: Format string from Progmem and print it

ricreis394, 2021-03-21:
Hello,

I'm using your library and I'm enjoying it so far. Thanks!

I have a question, I have a string stored in progmem, what's the best way to format it and print to the webserver?
I tried with `server.printf()` but seems to be limited to 128 chars

unwiredben, 2021-03-22:
This is a limitation of the Arduino library.  Try doing the formatting as a series of prints instead of trying to build a string in RAM and output it.  This was designed for a low-RAM 8-bit processor back in 2010, and I certainly don't recommend using this library for anything these days with much better web server systems available on hardware like the ESP32 or Raspberry Pi.

closed 2021-03-22 by unwiredben

ricreis394, 2021-03-22:
I'll be using Arduino MEGA, so it's not like ESP32 hardware.
I really liked your library and it's not much oversized.
Anyways, which library would you recommend instead of yours?